lnls-dig/openMMC #9: IPMI over RMCP bug

qermit, 2015-09-04:
ipmitool it will always send message with sequence number 7.
It seems sequence number is not mapped.
Code will process only first message and later ignore next incoming messages.

Tested with N.A.T. MCH:
    SESSION(0): closing session 0x00000005
    SESSION(0): activated session 0x00000005
    R(7,3,1)R(7,3,2)ipmi_SendFru(7): timeout - no response for REQ: 0x20->0x76, Seq=7 GET_DEVICE_ID_REQ
    WARN - SMQ(0x81->0x20->0x82:0): SEND_MESSAGE_REQ - timeout remove
    WARN - SMQ(0x20->0x82->0x76:1): GET_DEVICE_ID_REQ - timeout remove
    ERR - SMQ(0x82->0x20): SEND_MESSAGE_RSP - unexpected response
    ERR - SEND_MSG_RSP(0x82->0x20): no outstanding message
    SESSION(0): closing session 0x00000005
    SESSION(0): activated session 0x00000005
    R(7,3,1)R(7,3,2)ipmi_SendFru(7): timeout - no response for REQ: 0x20->0x76, Seq=7   GET_DEVICE_ID_REQ
    WARN - SMQ(0x81->0x20->0x82:0): SEND_MESSAGE_REQ - timeout remove
    WARN - SMQ(0x20->0x82->0x76:1): GET_DEVICE_ID_REQ - timeout remove
    ERR - SMQ(0x82->0x20): SEND_MESSAGE_RSP - unexpected response
    ERR - SEND_MSG_RSP(0x82->0x20): no outstanding message
    SESSION(0): closing session 0x00000005
    SESSION(0): activated session 0x00000005
    R(7,3,1)R(7,3,2)ipmi_SendFru(7): timeout - no response for REQ: 0x20->0x76, Seq=7 COLD_RESET_REQ
    WARN - SMQ(0x81->0x20->0x82:0): SEND_MESSAGE_REQ - timeout remove
    WARN - SMQ(0x20->0x82->0x76:1): COLD_RESET_REQ - timeout remove
    ERR - SMQ(0x82->0x20): SEND_MESSAGE_RSP - unexpected response
    ERR - SEND_MSG_RSP(0x82->0x20): no outstanding message
    SESSION(0): closing session 0x00000005

henrique-silva, 2015-09-04:
Yes, according to the IPMB specification: 
"The Requester must change the Seq field value for each new instance of a request that has the same
rsSA, rsLUN, netFn, and cmd fields as a previous request to the node (within the Seq field expiration
interval, see Section 2.6.2, Seq Value Expiration). The value can be repeated among outstanding
requests as long as at least one of the other fields is different."
In this case, if we want to check the `seq` field, we must compare also `rsSA`, `rsLUN`, `netFn`, and `cmd` fields

qermit, 2015-09-04:
This will not solve problem.
Each time I sent IPMI message over LAN the **seq** and **rsLUN** stay the same.
N.A.T. MCH waits until  processing of the first IPMI message finishes and sends next message with the same **seq** and **rsLUN**.
I created a followed scenario:
* Two RMCP sessions are created in parallel.
* MMC does not respond once every 3 messages
* Two the same commands are sent (0x06/0x01)

Here is result:

    R(7,3,1)SESSION(0): closing session 0x00000005
    SESSION(0): activated session 0x00000005
    SESSION(0): closing session 0x00000005
    SESSION(0): activated session 0x00000005
    SESSION(0): closing session 0x00000005
    SESSION(0): activated session 0x00000005
    R(7,3,1)SESSION(0): closing session 0x00000005
    SESSION(0): activated session 0x00000005
    SESSION(0): closing session 0x00000005
    SESSION(0): activated session 0x00000005
    SESSION(1): activated session 0x00000006
    SESSION(0): closing session 0x00000005
    R(7,3,1)SESSION(1): closing session 0x00000006
    SESSION(0): activated session 0x00000005
    SESSION(1): activated session 0x00000006
    SESSION(0): closing session 0x00000005
    SESSION(1): closing session 0x00000006
    SESSION(0): activated session 0x00000005
    SESSION(1): activated session 0x00000006
    R(7,3,1)SESSION(0): closing session 0x00000005
    SESSION(1): closing session 0x00000006
    SESSION(0): activated session 0x00000005
    SESSION(1): activated session 0x00000006
    SESSION(0): closing session 0x00000005
    R(7,3,1)SESSION(1): closing session 0x00000006

aylons, 2015-09-04:
If all fields are the same, including sequential number, how could we solve it? Or should we?

I guess we could respond to repeated requests instead of dropping them. The IPMB standard, section 2.6.1 (page 17) says we can do both, but if dropping it is impairing functionality, I think we should respond them all. So we would not check if the request is repeated and simply feed the queue with them. However, this could result in duplicated side effects caused by the handler function, if care is not taken.

Alternatively, instead of dropping the request, we could immediately send a 0xCF completion code, "Cannot execute duplicated request", How does IPMI tool deal with it? This would also prevent unwanted side effects.

henrique-silva, 2015-09-04:
I think I've discovered why the firmware only fails to respond every third message.

We store the last received req in a local variable inside IPMB_RXTask, so, the first ipmitool (I'll call it IP1) sends its message to the AFC, and gets its response. At this time, last_received_req holds IP1 message.
Then, IP1 send the same message again (which I think is a wrong behaviour, since the message has the same rsSA, rsLUN, NetFN and cmd, but is not a retry, it should have a different seq), but when the MMC compares it to `last_received_req`, is sees that is the same request and interprets it as a retry, so it is not responded.
When IP2 is spawned, it send a different message (probably different seq?). The MMC receives it and execute the desired function. `last_received_req` is then updated to the IP2 message.

If IP1 then sends its request now, it will be responded properly, because it's a different seq value.

If you run the firmware with the seq checking commented, does this problem still happens?

qermit, 2015-09-06:
I just created a code to test MCH behavior. A messeges were dropped in output queue. When I modified a code to respond only every fourh message I never observed they were mixed together. Always only one was processed. Even if i mixed this RMCP messages with requests from MMC (show_sensorinfo). 

So I will repeat one more time **Only last one recieved request is valid**.

If second message was recieved from IPMB channel while MMC is still processing previous one, we may suspect a significant delay in our code and that's it. If MMC is not processing any other request request must be processed. No magic inside.
There are several reason to do repeated request (message was not delivered to MMC, MMC had some delay, a recieved response was not matching sent request)
MMC shouldn't try to correct MCH and do not check incomming messages aganst sequence id or send message in a style "I know better". Probably there was a realy good reason to send message message with this sequence ID. Just imagine, while debugging session you may stop CPU. It's possible that sequence id will roll over. Or something happend, one bit was changed so CRC from response is not valid anymore. You have to remember there is not response to response.  

BTW  this sequence ID = 7 was correct. There were additional messages to authenticate client. ipmitool was not sending request one more time, MCH (or last bridge) was requesting it one more time.

henrique-silva, 2015-09-18:
I've written a simple python script to create two parallel ipmitool sessions and test this issue with the lattest commit 46397dd18efb58b85181a027f3bb51a02d9c3a7a in which we have fixed the IPMI useless checking for responses
The result follows:
```
SESSION(0): activated session 0x00000005
TX CM->MMC: *** GET_DEVICE_ID_REQ ***
  src: rqAddr 0x20 LUN 2   dest:  rsAddr 0x7a LUN 0   rqSeq: 0x07
  CRC1: 0x6e (ok)   CRC2: 0xc1 (ok) 
  no message data

RX CM: *** GET_DEVICE_ID_RSP ***
  src: rsAddr 0x7a LUN 0   dest:  rqAddr 0x20 LUN 2   rqSeq: 0x07
  CRC1: 0xc2 (ok)   CRC2: 0x5a (ok) 
  completion code: NORM (0x00)
  message data (len 12):
  00 0a 02 05 50 02 1f 5a   31 00 01 01

SESSION(0): closing session 0x00000005
SESSION(0): activated session 0x00000005
SESSION(1): activated session 0x00000006
TX CM->MMC: *** GET_DEVICE_ID_REQ ***
  src: rqAddr 0x20 LUN 2   dest:  rsAddr 0x7a LUN 0   rqSeq: 0x07
  CRC1: 0x6e (ok)   CRC2: 0xc1 (ok) 
  no message data

RX CM: *** GET_DEVICE_ID_RSP ***
  src: rsAddr 0x7a LUN 0   dest:  rqAddr 0x20 LUN 2   rqSeq: 0x07
  CRC1: 0xc2 (ok)   CRC2: 0x5a (ok) 
  completion code: NORM (0x00)
  message data (len 12):
  00 0a 02 05 50 02 1f 5a   31 00 01 01

TX CM->MMC: *** GET_DEVICE_ID_REQ ***
  src: rqAddr 0x20 LUN 2   dest:  rsAddr 0x7a LUN 0   rqSeq: 0x07
  CRC1: 0x6e (ok)   CRC2: 0xc1 (ok) 
  no message data

SESSION(0): closing session 0x00000005
RX CM: *** GET_DEVICE_ID_RSP ***
  src: rsAddr 0x7a LUN 0   dest:  rqAddr 0x20 LUN 2   rqSeq: 0x07
  CRC1: 0xc2 (ok)   CRC2: 0x5a (ok) 
  completion code: NORM (0x00)
  message data (len 12):
  00 0a 02 05 50 02 1f 5a   31 00 01 01

SESSION(1): closing session 0x00000006
SESSION(0): activated session 0x00000005
TX CM->MMC: *** GET_DEVICE_ID_REQ ***
  src: rqAddr 0x20 LUN 2   dest:  rsAddr 0x7a LUN 0   rqSeq: 0x07
  CRC1: 0x6e (ok)   CRC2: 0xc1 (ok) 
  no message data

RX CM: *** GET_DEVICE_ID_RSP ***
  src: rsAddr 0x7a LUN 0   dest:  rqAddr 0x20 LUN 2   rqSeq: 0x07
  CRC1: 0xc2 (ok)   CRC2: 0x5a (ok) 
  completion code: NORM (0x00)
  message data (len 12):
  00 0a 02 05 50 02 1f 5a   31 00 01 01

SESSION(0): closing session 0x00000005

```

As we can see, all messages that were bridged to  the MMC got their responses, there's no timeout like this one below, indicating that the MMC failed to respond a request
```
R(7,3,1)R(7,3,2)ipmi_SendFru(7): timeout - no response for REQ: 0x20->0x76, Seq=7 GET_DEVICE_ID_REQ
```

closed 2016-05-31 by henrique-silva