lnls-dig/openMMC #16: Chip_I2C_MasterTransfer hangs out forever if the slave doesn't respond

henrique-silva, 2015-10-01:
This mostly happens when the selected address is wrong, making the LPCOpen function hangs in the following loop:
```
while (Chip_I2C_MasterTransfer(id, &xfer) == I2C_STATUS_ARBLOST) {}
```

qermit, 2015-10-02:
Not a timeout but correct i2c bus handling, In this case NAK (SLA+W/SLA+R) should be triggered.
xfer.status = I2C_STATUS_NAK  will be returned and xfer.txSz should be not changed.

A code may hang in this state only if i2c interface has no pullup resistors or something is disturbing lines.

augustofg, 2025-04-24:
This only happens when the I2C lines are held LOW for some motive (lack of pull-ups, slave not powered, short-circuit), so this isn't a problem with LPCOpen. A timeout would be great to indicate a failure of the I2C bus, but it would be a non recoverable problem anyway.

closed 2025-04-24 by augustofg