lnls-dig/openMMC #49: Create new builds for different boards, not different versions

henrique-silva, 2016-07-26:
As for now, we create a different build for each board (and each version of it) that the openMMC is ported to. 
@lerwys suggested that we included the version differentiation in the firmware itself, so we have a single port for a type of board.

Palmitoxico, 2021-02-12:
I think we discussed this previously and converged in to keeping different builds for each board. Unifying all ports into a single binary would require significant architecture changes and would result in a fat binary that may not fit into the LPC1764's flash (the effective maximum size of application firmware is only 60KiB).

@lerwys do you remember this discussion?

lerwys, 2021-02-12:
Hum... I remember two things.

The first one was to use the board version id (e.g., v3.0, v3.1 for the afc-bpm) from the EEPROM. We already store that information in the FRU. One example: https://github.com/lnls-dig/openMMC/blob/master/port/board/afc-bpm/v3_1/user_amc_fru.h#L45 or https://github.com/lnls-dig/openMMC/blob/master/port/board/afc-bpm/v3_1/user_amc_fru.h#L46

With this version ID we could "merge" the builds for different versions (e.g., afc-bpm v3.0 and afc-bpm v3.1) to a single place and generate a single binary for all of them. Maybe this could help maintenance int he long term. At least as of today, if we want to fix or enhance something we need to implement that for all versions of that board.

As the ports for different version of the same board don't seem too different, it shouldn't increase the binary size too much, I guess.

The other idea was to "merge" different variants of the same board (e.g., afc-bpm and afc-timing) into the same binary. Again we already store that info in the FRU. Example: https://github.com/lnls-dig/openMMC/blob/master/port/board/afc-bpm/v3_1/user_amc_fru.h#L44

As the differences between the variants are also small this could improve maintainability and avoid copy-paste code.

What do you think?

lerwys, 2021-02-12:
Another idea that I've just remembered.

Instead of trying to do this dynamic board version/variant discovery we could just have the common parts of all AFC "ports" into a single location and keeping only what's actually different in different files.

This wouldn't involve any changes to the generated binary, only code reorganization.

lerwys, 2021-02-13:
Related to #96

Palmitoxico, 2021-02-15:
If the differences between the bpm and timing, 3.0 and 3.1 are small enough, I would prefer merging in a single binary, lump everything into an afcv3 common code base.

lerwys, 2021-02-15:
Nice! Sound good!

henrique-silva, 2021-02-16:
Hi guys,

The reason that we had two different ports for AFC and AFC-Timing was because in the early days of AFC v3.0 and development of v3.1 the hardware was a bit different between timing and BPM boards and the timing board was still under heavy debug, so I separated the builds to avoid bringing unwanted bugs to the more stable AFC used for BPMs.

If I recall correctly, the differences between them now are mostly on clock configuration, everything else is probably the same. I'd only confirm if the pin assignments are the same, just to be sure.

> If the differences between the bpm and timing, 3.0 and 3.1 are small enough, I would prefer merging in a single binary, lump everything into an afcv3 common code base.

Yes, this is a great idea, it should simplify a lot the maintenance.

augustofg, 2023-10-04:
We deprecated AFCv3.0 support, we don't use it, I'm not aware of anyone that uses it, so keeping it would bring an unnecessary burden to us. AFCv3.1 Timing and BPM support where merged into a unified binary now that it is possible to configure the clock switch via IPMI.

closed 2023-10-04 by augustofg