lnls-dig/openMMC #67: Increase LPC1764 working frequency

henrique-silva, 2017-03-20:
This is related to #39 
With higher core frequencies we can achieve higher baud rates in the Serial port. Also, we won't stall when too many IPMI messages come in sequence, given that we'll be fast enough to respond all of them.

I've performed some test with the AFC board and there seems to be no significant increase in the current draw by the LPC17xx working at 100MHz (max frequency for this part number - LPC1769 can reach 120MHz)

henrique-silva, 2017-04-05:
Increasing the working frequency to 100MHz causes 4 messages to not be responded to the MCH.
Couldn't exactly figure out with which of them this is happening because when the IPMI debug is enabled on MCH, the transactions ocurr very slowly and all of them are responded correctly.
The problem may be on the blocking inside IPMB task.

qermit, 2017-04-06:
I would suggest to connect external i2c sniffer and use wireshak. I did it in the past.