lnls-dig/openMMC #74: Using the I2C bus via FPGA causes the AFC/AFCK to reset

abyszuk, 2017-04-27:
I've noticed that it's basically impossible to control I2C via FPGA on AFCv3.1/AFCK.
If I try e.g. to change ADN4604 settings, all voltages on board will reset almost immediately.
I didn't check yet what exactly causes this bug, but my current suspect is that it's connected to PCA9547 I2C mux.
Maybe openMMC doesn't track the mux settings, so if the mux output changes silently (from LPC point of view) it triggers a bug in it?

henrique-silva, 2017-05-15:
When you say all voltages, you mean all the Payload DCDCs (red leds)?
If yes, then probably openMMC is having a HardFault and the watchdog module is resetting the chip. But I don't know what could cause this bug, maybe a invalid pointer or something alike.

Can you post how do you configure the mux? I'll try to reproduce this problem here so I can investigate.

> Maybe openMMC doesn't track the mux settings, so if the mux output changes silently (from LPC point of view) it triggers a bug in it?

The firmware actually checks the mux state, but just when a task wants to take the I2C semaphore. So if one would transfer a long list of bytes while a task is holding the semaphore, the mux could change in this interval without the openMMC noticing.

abyszuk, 2017-05-30:
Yes, all payload DCDCs go off.

Basically, the code which can trigger this problem is:
```
glc_pca9547_switch_output(&pca_main, 4);
//do some write/read cycles to test ADN4604 I2C bus
for(int i = 7; i >= 0; i--) {
  glc_write(&adn_xbar, 0x98 + i, i);
  glc_read(&adn_xbar, 0x98 + i, &adn_read);
}
```
Which I2C bus is used by openMMC? Do you always use FPGA_I2C, or do you use I2C_CPU when possible?

Palmitoxico, 2022-01-04:
The way the I2C buses are implemented in AFCv3.1 is a mess... The best you can do is either give the MMC exclusive access to the PCA9547 mux or give it to the FPGA. If you don't care about RTM support you can leave the PCA9547 mux to the FPGA and use the MMC I2C1 (P0.0, P0.1) bus to access the voltage / current / temperature sensors.

[image]

This is a hardware bug that was fixed in the AFCv4 revision by using different muxes for the MMC and FPGA.

closed 2022-01-05 by Palmitoxico