lnls-dig/openMMC #79: Use openOCD as an alternative to program the LPC controller

henrique-silva, 2017-09-01:
We should avoid using commercial software to program our board, since it limits us in terms of flexibility.
A great option is to use openOCD as an interface working together with a USB-Serial converter (FTDI, Buspirate, etc) to program the chip.

jordens, 2017-12-17:
For reference, on Sayma-AMC we have been looking into this a bit: https://github.com/m-labs/sinara/issues/448

sbourdeauducq, 2017-12-18:
It pretty much works out of the box, except for standard fare JTAG connection flimsiness. Here's the script I'm using:
```
source /usr/share/openocd/scripts/interface/ftdi/olimex-arm-usb-tiny-h.cfg
source /usr/share/openocd/scripts/target/lpc17xx.cfg
adapter_khz 100

program Downloads/lpc_1776_ethernet.axf verify
reset
exit
```

commit 3f684a10 (2021-06-09): Migrate to OpenOCD

Use OpenOCD for flashing and debugging instead of proprietary NXP
tools. For LPC-Link2 probes you must flash the cmsis-dap or jlink
firmware.

LPC-Link1 probes aren't supported by OpenOCD, you will be required to
install the MCUExpresso IDE and use the probe/lpclink1-flash.sh
script.

Closes #79
closed 2021-06-09 by Palmitoxico