lnls-dig/openMMC #83: Use FreeRTOS software timers to schedule resets

henrique-silva, 2018-01-31:
Using this feature from FreeRTOS would allow us to decouple the scheduled resets (needed in order to have time to respond IPMI messages before rebooting) from some modules like the watchdog.

henrique-silva, 2018-01-31:
This would cause a larger memory usage, since the FreeRTOS timers are managed by a dedicated background task. If there are any other uses for timers in the firmware, this could be a good approach.
Other solution would be to use LPC internal timers to do this, but this could conflict with other (future) modules or ports, since timers are vastly used like in PWM, motor control, etc.

henrique-silva, 2018-01-31:
We could also use a software timer to perform the whole watchdog feeding proccess

Palmitoxico, 2021-04-27:
Done in aa33284e4054c4d60928e5896d7f53ad1366568a

closed 2021-04-27 by Palmitoxico