lnls-dig/openMMC #86: Test build on Windows

henrique-silva, 2018-03-26:
There should be some problems with the libraries and headers paths when compiling the firmware on Windows.

paultech, 2020-01-24:
Builds fine under Windows with newest (9-2019-q4-major) GNU-Arm toolchain

`make
[ 11%] Built target FreeRTOS
[ 42%] Built target lpcopen
[ 94%] Built target openMMC
[100%] Built target bootloader
`

`file build/out/*
build/out/bootloader.axf:                                                          ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), statically linked, not stripped
build/out/bootloader.bin:                                                          data
build/out/bootloader_pad.bin:                                                      data
build/out/C:Userspaultechscoopappsqemucurrentlibgdk-pixbuf-2.02.10.0loaders.cache: empty
build/out/openMMC.axf:                                                             ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), statically linked, not stripped
build/out/openMMC.bin:                                                             data
build/out/openMMC_full.bin:                                                        data`

augustofg, 2024-08-22:
I don't think openMMC builds on Windows anymore (due some Unix commands necessary to generate the combined bootloader + application binary). If someone want to support building on other platforms, PRs are welcome.

closed 2024-08-22 by augustofg