lnls-dig/openMMC #92: event_receiver_addr should be MCH_ADDRESS

umairsiddiqui-digitek, 2018-08-09:
https://github.com/lnls-dig/openMMC/blob/f11f542f4ac210cefd857bb082a86c410e434fba/modules/sdr.c#L236

this line should be 

`event_receiver_addr = MCH_ADDRESS;`

btw this variable is only used in `get_event_receiver` handler.

henrique-silva, 2018-09-21:
It could be, but I left it as unspecified on purpose, as the IPMI standard requires the MCH to send a `Set Event Receiver` request to the MMCs upon initialization. 
This way we only send events if the MCH actively requested us to do it (by setting the event receiver address)

henrique-silva, 2019-01-24:
On second though, it's safer to just set this address before than depending on the MCH to actively set it.
Fixed on 6a6725f9e27a8b794b00ac63405ffe1b1468f5ca

closed 2019-01-24 by henrique-silva