lnls-dig/openMMC #93: the ownerid in sdr_list not updated

umairsiddiqui-digitek, 2018-08-13:
the ownerid in sdr_list not updated because `get_ipmb_addr()` is called after `sdr_init()`.

https://github.com/lnls-dig/openMMC/blob/f11f542f4ac210cefd857bb082a86c410e434fba/modules/main.c#L85

i think correct sequence should be 

```
fru_init();
ipmi_init();
sdr_init();
```

henrique-silva, 2018-09-21:
The only problem I see is that if we call ipmi_init before the other initializing functions, some I2C interruptions may happen on the meantime, which could cause some problems.
Maybe the best option is to split the `get_ipmb_addr` function from the `ipmi_init` function and call it before everything else (this is just a pin reading and some math, so nothing to worry here).

henrique-silva, 2019-01-24:
This was causing problems on sensor's reading (see #98) and was fixed on 65012d8839decd21afca57699a934454ff05d84b

closed 2019-01-24 by henrique-silva