lnls-dig/openMMC #123: Configure clock switch via IPMI and save it to the FRU EEPROM

Palmitoxico, 2022-03-07:
The clock switch configuration shouldn't be hard-coded to the MMC firmware. Instead, it should be loaded from the EEPROM and configured via IPMI. The AFCv3.1 port supports changing the adn4604 configuration via IPMI, but it doesn't expose a way to read the current configuration and store it in the EEPROM.

If possible, design an agnostic IPMI message API that can be used for adn4604 and idt_8v54816.

augustofg, 2023-08-02:
For the IPMI / EEPROM API I propose using the 8V54816ANLG register layout as it is very simple and direct to the point, then for the AFCv3.1 translate it to something that makes sense:
[image]

augustofg, 2023-08-02:
So, for the AFCv3.1 (ADN4604ASVZ) boards I've come with the following scheme:
- **Port I/O (bit 7)**: Ignored, input and outputs are fixed;
- **Termination On/Off (bit 6)**:  Ignored, ADN4604ASVZ doesn't have individual port termination control (one bit controls 8 ports). All inputs will have its terminators ON;
- **Polarity (bit 5)**: ADN4604ASVZ only allow changing the polarity of input ports, so this will only apply its inputs;
- **Output Port Signal Source (bits 0 to 3)**: Select the input port for the respective output port.

closed 2023-09-29 by gustavosr8