lnls-dig/openMMC #131: Simplify the payload power state machine, add proper support to all 'FRU Control Commands'

Palmitoxico, 2023-03-17:
The current `vTaskPayload` task is hard to understand and doesn't implement all FRU Control Options in a usable way. We should decide what each command will do, trying to conform with the AdvancedTCA standard.

The FRU Control Command has the following options:
```
0x00 Cold Reset
0x01 Warm Reset
0x02 Graceful Reboot
0x03 Issue a Diagnostic Interrupt
```

Please check the PICMG 3.0 R3.0 AdvancedTCA Base Specification pages 3-60, 3-61 and 3-62.

Palmitoxico, 2023-03-17:
This can be done at the same time as #130

augustofg, 2023-05-18:
May be it would be better to also check PICMG Advanced Mezzanine Card AMC.0 Specification R2.0, pages 3-75 and 3-76.